chestm007/amdgpu-fan #1: Options for selecting GPU(s) in /etc/amdgpu-fan.yml

DerVerruckteFuchs, 2018-12-22:
Is there a means for running amdgpu-fan only with (a) selected GPU(s)? I have my Intel iGPU enabled and it's `/sys/class/drm/card0` whereas my AMD GPU is `/sys/class/drm/card1`. It seems `amdgpu.py` looks for `card0` by default. It looks like I could use `amdgpu.py --get card1 fan_speed` from the help flag, but It only looks for `card0`. My Intel iGPU does not seem to have a `/device/hwmon` directory, so `amdgpu.py` fails with the following traceback:

`Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/amdgpu.py", line 113, in <module>
    scanner = Scanner()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/amdgpu.py", line 69, in __init__
    self.cards = self._get_cards()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/amdgpu.py", line 76, in _get_cards
    return {node: Card(node) for node in os.listdir(ROOT_DIR) if re.match(self.CARD_REGEX, node)}
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/amdgpu.py", line 76, in <dictcomp>
    return {node: Card(node) for node in os.listdir(ROOT_DIR) if re.match(self.CARD_REGEX, node)}
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/amdgpu.py", line 15, in __init__
    for node in os.listdir(os.path.join(ROOT_DIR, self._identifier, HWMON_DIR)):
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/drm/card0/device/hwmon'
`

DerVerruckteFuchs, 2018-12-22:
I looked closer at `amdgpu.py`, and it looks like it should always fail should `/sys/class/drm/cardX/device/hwmon` be missing for any card. I think checking for `hwmon` directories should be performed first, then only running the code at line 15 for cards that have an existing `hwmon` directory should prevent `amdgpu.py` from failing.

chestm007, 2019-02-08:
from a quick look it seems i should put some code to handle non-amd gpu's in the same system - it appears this code makes a blanket assumption that all cards in the system are AMD or atleast will have hwmon.

I should be able to get to this soon

chestm007, 2019-02-08:
@DerVerruckteFuchs so i have an implementation of this in master currently (0.0.2+) please let me know if it fixes your problem.

edacval, 2019-02-08:
> @DerVerruckteFuchs so i have an implementation of this in master currently (0.0.2+) please let me know if it fixes your problem.

Thanks, works in dual gpu system.

chestm007, 2019-02-08:
cool, I'll leave this open however as I'm sure that being able to specify the GPU's you'd like this to act on might actually be useful to some people and seems like something i should implement.

chestm007, 2019-03-07:
selectable cards implemented in 7184fb6b060a8687c447d567a257c4d14b7a256e

closed 2019-03-07 by chestm007