chestm007/amdgpu-fan #2: undeclared dependencies

FelixSchwarz, 2018-12-30:
As far as I can see `setup.py` does not declare any dependencies. However the scripts needs at least PyYAML, numpy, matplotlib and scipy. (Also I'd appreciate if it could run without the later three.)

chestm007, 2019-02-08:
The later 3 are required for the fan curve logic - I'm open to accepting a PR that can replace that functionality without using those libraries.

As far as the dependencies - I'll setup a clean environment this afternoon and sort that out

chestm007, 2019-02-08:
this should be fixed in the latest master (0.0.2+)

chestm007, 2019-03-07:
@FelixSchwarz ive removed matplotlib and scipy, numpy has to stay as it does the processing logic for the fan curve still

These are in effect as of 0.0.3

closed 2019-03-07 by chestm007