chestm007/amdgpu-fan #3: AttributeError: 'list' object has no attribute 'get'

YourHuckleberry, 2019-01-17:
Unable to run (installed from AUR).

`$ sudo amdgpu-fan
Traceback (most recent call last):
  File "/usr/bin/amdgpu-fan", line 11, in <module>
    load_entry_point('amdgpu-fan===0.0.1.r1.g6d29514', 'console_scripts', 'amdgpu-fan')()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 58, in main
    FanController(config).main()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 19, in __init__
    self.curve = Curve(config.get('speed_matrix'))
AttributeError: 'list' object has no attribute 'get'
`

Config file is at /etc/amdgpu-fan.yml and is 777.  Contents exactly same as in readme.

chestm007, 2019-02-08:
it looks like the config in the readme is actually wrong. I'll take a closer look this afternoon

chestm007, 2019-02-08:
fixed in 50d74d247793e825a2672fb8f7cd9530241b5241 i believe
the problem was the example was missing the `speed_matrix:` line. sorry.

chestm007, 2019-02-08:
this should be fixed in the latest master (0.0.2+)

closed 2019-03-07 by chestm007