chestm007/amdgpu-fan #5: amdgpu-fan crashing

edacval, 2019-02-08:
```
Traceback (most recent call last):
  File "/usr/bin/amdgpu-fan", line 11, in <module>
    load_entry_point('amdgpu-fan==0.0.2', 'console_scripts', 'amdgpu-fan')()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 70, in main
    FanController(config).main()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 31, in main
    logger.debug(f'{name}: Temp {temp}, Setting fan speed to: {speed}, fan speed{card.fan_speed}, min:{card.fan_min}, max:{card.fan_max}')
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/amdgpu.py", line 38, in fan_speed
    return int(self.read_endpoint('fan1_input'))
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/amdgpu.py", line 29, in read_endpoint
    with open(self._endpoints[endpoint], 'r') as e:
KeyError: 'fan1_input'
```
[https://github.com/chestm007/amdgpu-fan/blob/master/amdgpu_fan/lib/amdgpu.py#L38](url) tries to open  /sys/class/drm/card0/device/hwmon/hwmon0/fan1_input, but no such file exists.
```
ls -Fh /sys/class/drm/card0/device/hwmon/hwmon0/:
device@  name  power/  pwm1  pwm1_enable  pwm1_max  pwm1_min  subsystem@  temp1_crit  temp1_crit_hyst  temp1_input  uevent
```
Applied temporary workaround, but not sure if it is right:
```
diff --git a/amdgpu_fan/lib/amdgpu.py b/amdgpu_fan/lib/amdgpu.py
--- a/amdgpu_fan/lib/amdgpu.py
+++ b/amdgpu_fan/lib/amdgpu.py
@@ -35,7 +35,7 @@ class Card:
 
     @property
     def fan_speed(self):
-        return int(self.read_endpoint('fan1_input'))
+        return int(self.read_endpoint('pwm1'))
 
     @property
     def gpu_temp(self): 
```
Up to date Arch Linux, 4.19.20-1-lts kernel, Radeon R7 370

chestm007, 2019-02-08:
Thats interesting, It appears that your card doesn't support reading the current fan speed, I think in this case its better to sanely return the value as None rather then explode in the users face as it apparently does currently, I should have a chance to look into this further this weekend.

chestm007, 2019-03-07:
@edacval I believe this issue should be fixed in 0.0.3, please let me know if im wrong

edacval, 2019-03-07:
Error changed to:
```
Mar 07 14:20:35 amd amdgpu-fan[9685]: starting amdgpu-fan
Mar 07 14:20:35 amd amdgpu-fan[9685]: Traceback (most recent call last):
Mar 07 14:20:35 amd amdgpu-fan[9685]:   File "/usr/bin/amdgpu-fan", line 11, in <module>
Mar 07 14:20:35 amd amdgpu-fan[9685]:     load_entry_point('amdgpu-fan==0.0.4', 'console_scripts', 'amdgpu-fan')()
Mar 07 14:20:35 amd amdgpu-fan[9685]:   File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 78, in main
Mar 07 14:20:35 amd amdgpu-fan[9685]:     FanController(config).main()
Mar 07 14:20:35 amd amdgpu-fan[9685]:   File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 35, in main
Mar 07 14:20:35 amd amdgpu-fan[9685]:     logger.debug(f'{name}: Temp {temp}, Setting fan speed to: {speed}, fan speed{card.fan_speed}, min:{card.fan_min}, max:{card.fan_max}')
Mar 07 14:20:35 amd amdgpu-fan[9685]:   File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/amdgpu.py", line 53, in fan_speed
Mar 07 14:20:35 amd amdgpu-fan[9685]:     return int(self.read_endpoint('fan1_input'))
Mar 07 14:20:35 amd amdgpu-fan[9685]: TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
Mar 07 14:20:35 amd systemd[1]: amdgpu-fan.service: Main process exited, code=exited, status=1/FAILURE
Mar 07 14:20:35 amd systemd[1]: amdgpu-fan.service: Failed with result 'exit-code'.

```

chestm007, 2019-03-07:
:man_facepalming: I derped, should work on any commit after 80ac3ffa8bbfb6d74aa557fffdacb91902ee6109 now

edacval, 2019-03-08:
:joy: It works !

closed 2019-03-11 by edacval