chestm007/amdgpu-fan #7: execute via systemctl rather than directly via the binary

aumars, 2019-04-23:
I can execute the service via `systemctl` rather than executing `amdgpu-fan`. Maybe you should reflect that in the `README`.

chestm007, 2019-04-23:
Yeah, that's the way i run this also - I was contemplating adding a systemd unit to this repository and providing install instructions but thought it might be best to leave that up to people implementing distro packages.

Happy to add systemd unit files and instructions to the readme if you think its worth it however

vale981, 2019-09-08:
It wasn't to hard to create one yourself, but a hint would be handy.

FKleinebreil, 2019-10-28:
I'm new to Linux and would like to make the fan curve persist upon reboot. I put the `amdgpu-fan.service` in `/lib/systemd/system/`, but form my understanding I also have to put a file named `amdgpu-fan` in `usr/bin`, correct? What needs to be written in that file, or where can I find it? Hin in the `README` would indeed be nice. Very handy script otherwise!

psy-q, 2020-04-17:
I'm using the most primitive systemd unit file on openSUSE (taken from Arch's AUR):

```
[Unit]
Description=amdgpu fan controller

[Service]
ExecStart=/usr/local/bin/amdgpu-fan
Restart=always

[Install]
WantedBy=default.target
```

`pip3 install` already gave me the right binary in /usr/local/bin/amdgpu-fan. Copy the unit file to `/etc/systemd/system/amdgpu-fan.service`, then you can enable with `systemctl enable amdgpu-fan.service` or start immediately with `systemctl start amdgpu-fan.service`.

closed 2025-02-28 by chestm007