chestm007/amdgpu-fan #10: amdgpu-fan does not appear to be running

Last-Son, 2019-05-10:
Running sudo amdgpu-fan from terminal results in this error:
starting amdgpu-fan
Traceback (most recent call last):
  File "/usr/bin/amdgpu-fan", line 11, in <module>
    load_entry_point('amdgpu-fan==0.0.5', 'console_scripts', 'amdgpu-fan')()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 78, in main
    FanController(config).main()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 35, in main
    logger.debug(f'{name}: Temp {temp}, Setting fan speed to: {speed}, fan speed{card.fan_speed}, min:{card.fan_min}, max:{card.fan_max}')
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/amdgpu.py", line 50, in fan_speed
    return int(self.read_endpoint('fan1_input'))
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/amdgpu.py", line 37, in read_endpoint
    return e.read()
OSError: [Errno 19] No such device

Checking systemctl --failed shows that the service is loaded but failed.

Am I doing something wrong?

Sorenwise, 2019-05-25:
Anyone got any ides for this I got this when I did amdgpu-fan --status and I have turned it on and it does nothing because I play a game and I saw the temp go to 70c and I am guessing it kept on going up till the computer crashed.

Traceback (most recent call last):
  File "/usr/bin/amdgpu-fan", line 11, in <module>
    load_entry_point('amdgpu-fan==0.0.5', 'console_scripts', 'amdgpu-fan')()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 78, in main
    FanController(config).main()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 23, in __init__
    self.curve = Curve(config.get('speed_matrix'))
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/curve.py", line 10, in __init__
    self.temps = self.points[:, 0]
IndexError: too many indices for array

chestm007, 2019-07-23:
@Last-Son `OSError: [Errno 19] No such device` hmm, which card are you running?

@Sorenwise what does your config file look like?

MasterCATZ, 2019-08-03:
aio@aio:~/Programs/amdgpu-fan$ sudo amdgpu-fan
Traceback (most recent call last):
  File "/usr/local/bin/amdgpu-fan", line 11, in <module>
    load_entry_point('amdgpu-fan==0.0.22.post1556022696', 'console_scripts', 'amdgpu-fan')()
  File "/usr/local/lib/python3.6/dist-packages/amdgpu_fan/controller.py", line 67, in main
    config = load_config(location)
  File "/usr/local/lib/python3.6/dist-packages/amdgpu_fan/controller.py", line 44, in load_config
    return yaml.load(f)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 72, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 35, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 439, in parse_block_mapping_key
    "expected <block end>, but found %r" % token.id, token.start_mark)
yaml.parser.ParserError: while parsing a block mapping
  in "/etc/amdgpu-fan.yml", line 2, column 1
expected <block end>, but found '<block sequence start>'
  in "/etc/amdgpu-fan.yml", line 14, column 2

MasterCATZ, 2019-11-05:
aio@aio:~/Programs/amdgpu-fan$ sudo amdgpu-fan
/home/aio/.local/lib/python3.6/site-packages/amdgpu_fan/controller.py:44: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  return yaml.load(f)
Traceback (most recent call last):
  File "/usr/local/bin/amdgpu-fan", line 11, in <module>
    load_entry_point('amdgpu-fan==0.0.22.post1556022696', 'console_scripts', 'amdgpu-fan')()
  File "/home/aio/.local/lib/python3.6/site-packages/amdgpu_fan/controller.py", line 78, in main
    FanController(config).main()
  File "/home/aio/.local/lib/python3.6/site-packages/amdgpu_fan/controller.py", line 23, in __init__
    self.curve = Curve(config.get('speed_matrix'))
  File "/home/aio/.local/lib/python3.6/site-packages/amdgpu_fan/lib/curve.py", line 10, in __init__
    self.temps = self.points[:, 0]
IndexError: too many indices for array

closed 2025-02-28 by chestm007