chestm007/amdgpu-fan #12: IndexError: too many indices for array

CypherNL, 2019-08-29:
OS: Arch Linux
GPU: MSI Radeon R9 390 Gaming (card0)

After the program creats a default config and uncommenting card0, the program outputs this:
```
/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py:44: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  return yaml.load(f)
Traceback (most recent call last):
  File "/usr/bin/amdgpu-fan", line 11, in <module>
    load_entry_point('amdgpu-fan==0.0.6', 'console_scripts', 'amdgpu-fan')()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 78, in main
    FanController(config).main()
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/controller.py", line 23, in __init__
    self.curve = Curve(config.get('speed_matrix'))
  File "/usr/lib/python3.7/site-packages/amdgpu_fan/lib/curve.py", line 10, in __init__
    self.temps = self.points[:, 0]
IndexError: too many indices for array
```

Config:
```
#Fan Control Matrix. [<Temp in C>,<Fanspeed in %>]
speed_matrix:
- [0, 0]
- [30, 33]
- [45, 50]
- [60, 66]
- [65, 69]
- [70, 75]
- [75, 89]
- [80, 100]

# optional
# cards:  # can be any card returned from `ls /sys/class/drm | grep "^card[[:digit:]]$"`
- card0
```
Any ideas what could be wrong?

chestm007, 2019-10-15:
you need to uncomment `cards:` also just above ` - card0`

CypherNL, 2019-10-27:
Whoops. Thanks for letting me know!

closed 2019-10-27 by CypherNL