chestm007/amdgpu-fan #14: Had to make an edit to get the service to run properly

ryenigma, 2019-11-22:
So I noticed today that my amdgpu-fan service wasn't running (found out due to an overheating issue in a game).  I had installed it using the pip3 installer and then to try and get the service running I downloaded the zip and moved the service file to /etc/systemd/system.  After enabling I hadn't taken the time to check but apparently it was actually failing to start.  Perhaps I am doing things wrong by combining installs but I had to edit the service file to point to /usr/local/bin instead of /usr/bin.  Not a huge thing but it did take a little tweaking.  Perhaps I am messing things up by downloading a separate service file but it is now working.  Just for reference I am basically using ubuntu 18.04.  With my tweak it seems to be working perfectly now.  Thanks btw way for such a nice tool for keeping things working.

duckduck1, 2019-11-30:
Not sure if these issues are related since files are in /usr/bin for me.

amdgpu-fan won't start anymore on up-to-date Arch Linux with recent python3 packages:

`journalctl -u amdgpu-fan.service`

> systemd[1]: Started amdgpu fan controller.
amdgpu-fan[1975]: Traceback (most recent call last):
amdgpu-fan[1975]:   File "/usr/bin/amdgpu-fan", line 6, in <module>
amdgpu-fan[1975]:     from pkg_resources import load_entry_point
amdgpu-fan[1975]:   File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3252, in <module>
amdgpu-fan[1975]:     def _initialize_master_working_set():
amdgpu-fan[1975]:   File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3235, in _call_aside
amdgpu-fan[1975]:     f(*args, **kwargs)
amdgpu-fan[1975]:   File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3264, in _initialize_master_working_set
amdgpu-fan[1975]:     working_set = WorkingSet._build_master()
amdgpu-fan[1975]:   File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 583, in _build_master
amdgpu-fan[1975]:     ws.require(__requires__)
amdgpu-fan[1975]:   File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 900, in require
amdgpu-fan[1975]:     needed = self.resolve(parse_requirements(requirements))
amdgpu-fan[1975]:   File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 786, in resolve
amdgpu-fan[1975]:     raise DistributionNotFound(req, requirers)
amdgpu-fan[1975]: pkg_resources.DistributionNotFound: The 'amdgpu-fan==0.0.6' distribution was not found and is required by the application
systemd[1]: amdgpu-fan.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: amdgpu-fan.service: Failed with result 'exit-code'.

hirowatari, 2020-12-04:
Thanks.  Also an Arch urser.  When updating Arch recently, this stopped working.  Reinstalling fixed it, but makes me unsettled.

for reference

```bash
$ amdgpu-fan
Traceback (most recent call last):
  File "/usr/bin/amdgpu-fan", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3239, in <module>
    def _initialize_master_working_set():
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3222, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 3251, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 567, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 884, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3.9/site-packages/pkg_resources/__init__.py", line 770, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'amdgpu-fan==0.0.6' distribution was not found and is required by the application
# now it's broken
$ sudo pacman -Rsn amdgpu-fan
$ yay amdgpu-fan
$ sudo systemctl restart amdgpu-fan.service
# now it's fixed
```

Presumably, this is a bug either in this project or in the AUR package.

chestm007, 2025-03-02:
This issue should be fixed, looks like it was about a year or so ago - apologies guys, i pretty much abandoned this a few years back to start a business. I'm back in the industry now and active again!

closed 2025-03-02 by chestm007