chestm007/amdgpu-fan #15: Doesn't appear to do anything

jp1995, 2020-01-22:
So I tried to get the advertised functionality out of the card, which is the fans not running under 60 degrees.

Config file:
```
#Fan Control Matrix. [<Temp in C>,<Fanspeed in %>]
speed_matrix:
- [0, 0]
- [30, 0]
- [45, 0]
- [60, 0]
- [61, 30]
- [65, 50]
- [70, 70]
- [75, 89]
- [80, 100]

# optional
cards:  # can be any card returned from `ls /sys/class/drm | grep "^card[[:digit:]]$"`
- card0
```
running amdgpu-fan as sudo results in: 
```
/usr/lib/python3.8/site-packages/amdgpu_fan/controller.py:44: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  return yaml.load(f)
starting amdgpu-fan
```
`watch sensors` reports fan speed steady at 1200 RPM. Visually the fans are still clearly running. Running amdgpu-fan as a service doesn't result in different behavior either. card0 exists in /sys/class/drm

GPU: AsRock RX 570 8GB
Distro: Manjaro

fuzunspm, 2021-01-22:
same issue here

psy-q, 2021-11-04:
It never manages to read the config file so it doesn't know what to do. This has been fixed by various people in different pull requests but the right fix depends on the version of PyYAML that gets installed.

I have a pull request open that also specifies the PyYAML version so it should fix this on modern systems: #23

If this doesn't get merged, you could try my fork via `python setup.py install`: https://github.com/psy-q/amdgpu-fan/tree/yaml_fix

closed 2025-02-28 by chestm007