chestm007/amdgpu-fan #16: This doesn't work at all on Ubuntu 19.10 and this project appears dead.

gardotd426, 2020-02-08:
When trying to run `amdgpu-fan` after installing it via `sudo pip3 install amdgpu-fan`, and then trying to run it, it throws a million errors:
```
sudo amdgpu-fan
/usr/local/lib/python3.7/dist-packages/amdgpu_fan/controller.py:44: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  return yaml.load(f)
Traceback (most recent call last):
  File "/usr/local/bin/amdgpu-fan", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/amdgpu_fan/controller.py", line 67, in main
    config = load_config(location)
  File "/usr/local/lib/python3.7/dist-packages/amdgpu_fan/controller.py", line 44, in load_config
    return yaml.load(f)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 41, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 439, in parse_block_mapping_key
    "expected <block end>, but found %r" % token.id, token.start_mark)
yaml.parser.ParserError: while parsing a block mapping
  in "/etc/amdgpu-fan.yml", line 2, column 1
expected <block end>, but found '<block mapping start>'
  in "/etc/amdgpu-fan.yml", line 19, column 2
```
I know the first error, the `/usr/local/lib/python3.7/dist-packages/amdgpu_fan/controller.py:44: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.` is not a big deal, as I have always gotten that error and it has worked on other systems (namely Arch Linux systems), but it's completely broken on Ubuntu. But then when I actually looked, I saw that this project is incredibly dead. It would be nice, if you had  no intention of fixing issues with your project, if you would at least say that the project is dead in the README, instead of people getting screwed over by installing a package that is listed in multiple places as being functional but is not, and if you have an issue there's no way to get it resolved.

psy-q, 2020-04-17:
Just dropping a report: I'm running this fine on openSUSE Tumbleweed today, and as that's a rolling distro it should be bleeding edge with amgpu, Python, etc. Running it on Python 3.8.2 with no errors or warnings besides the loader.

Blaimi, 2020-09-02:
> yaml.parser.ParserError: while parsing a block mapping
>   in "/etc/amdgpu-fan.yml", line 2, column 1
> expected <block end>, but found '<block mapping start>'
>   in "/etc/amdgpu-fan.yml", line 19, column 2

your config-file is corrupt

psy-q, 2021-11-04:
I take this back, depending on when you installed it, I guess you get different versions of PyYAML and the current code won't work on PyYAML > 5.1.0.

See #15.

chestm007, 2025-02-28:
that issue should be gone now, sorry ive been absent for an insanely long time :(

closed 2025-02-28 by chestm007