chestm007/amdgpu-fan #17: R9 390: fan1_input - No such device

kleest, 2020-02-25:
Using an R9 390, reading the current fan speed results in a program crash:
```
  File "/usr/lib/python3.8/site-packages/amdgpu_fan/controller.py", line 35, in main
    logger.debug(f'{name}: Temp {temp}, Setting fan speed to: {speed}, fan speed{card.fan_speed}, min:{card.fan_min}, max:{card.fan_max}')
  File "/usr/lib/python3.8/site-packages/amdgpu_fan/lib/amdgpu.py", line 50, in fan_speed
    return int(self.read_endpoint('fan1_input'))
  File "/usr/lib/python3.8/site-packages/amdgpu_fan/lib/amdgpu.py", line 37, in read_endpoint
    return e.read()
OSError: [Errno 19] No such device
```
This is due to `cat /sys/class/drm/card0/device/hwmon/hwmon1/fan1_input: No such device`

System information:
```
1d:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Hawaii PRO [Radeon R9 290/390] [1002:67b1] (rev 80)
Linux 5.5.5-arch1-1 #1 SMP PREEMPT Thu, 20 Feb 2020 18:23:09 +0000 x86_64 GNU/Linux
```

SpyrosRoum, 2020-04-20:
I am having the exact same issue, also with an R9 390 (from sapphire if that helps) though I find it weird because if I try to `ls /sys/class/drm/card0/device/hwmon/hwmon1/` I can see `fan1_input` listed

kleest, 2020-04-20:
For the time being, I fixed it with:
```diff
--- a/amdgpu_fan/lib/amdgpu.py
+++ b/amdgpu_fan/lib/amdgpu.py
@@ -48,7 +48,7 @@ class Card:
     def fan_speed(self):
         try:
             return int(self.read_endpoint('fan1_input'))
-        except KeyError:  # better to return no speed then explode
+        except:  # better to return no speed then explode
             return 0
 
     @property
```

Of course, this is no permanent solution.

SpyrosRoum, 2025-03-02:
Better late than ever. My r9 390 died a few years ago though so I wouldn't know

chestm007, 2025-03-02:
Damn, would have been nice to debug this - i switched from a RX480 to a watercooled 6800XT, so my need for this kinda went with it, didnt really expect it to get as much use from other people as it has