chestm007/amdgpu-fan #18: OSError: [Errno 22] Invalid argument

crazy-frag, 2020-08-06:
lsb_release -a
#Description:    Ubuntu 18.04.4 LTS

lspci | grep VGA
#01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere [Radeon RX 470/480/570/570X/580/580X] (rev ef)

python3 --version
#Python 3.6.9

sudo apt install python3-pip

pip3 --version
#pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.6)

sudo amdgpu-fan

starting amdgpu-fan
Traceback (most recent call last):
  File "/usr/local/bin/amdgpu-fan", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/amdgpu_fan/controller.py", line 7                                                8, in main
    FanController(config).main()
  File "/usr/local/lib/python3.6/dist-packages/amdgpu_fan/controller.py", line 3                                                5, in main
    logger.debug(f'{name}: Temp {temp}, Setting fan speed to: {speed}, fan speed                                                {card.fan_speed}, min:{card.fan_min}, max:{card.fan_max}')
  File "/usr/local/lib/python3.6/dist-packages/amdgpu_fan/lib/amdgpu.py", line 5                                                0, in fan_speed
    return int(self.read_endpoint('fan1_input'))
  File "/usr/local/lib/python3.6/dist-packages/amdgpu_fan/lib/amdgpu.py", line 3                                                7, in read_endpoint
    return e.read()
OSError: [Errno 22] Invalid argument

crazy-frag, 2020-08-07:
RX570, RX580

crazy-frag, 2020-08-07:
amdgpu-pro-20.30-1109583-ubuntu-18.04.tar.xz

apichlinski, 2020-08-24:
Hi,
I Installed newest official driver amdgpu-pro-20.30-1109583-ubuntu-20.04.tar.xz for Ubuntu 20.04 but without resolve.
Could you tell me what is wrong for RX580?

Best regards.

kluzas, 2021-01-16:
I had same error, and what I have figured out is that with default config in `/etc/amdgpu-fan.yml` it fails by reading temperature on my integrated GPU, which is `card0` on my system. So the solution is to add this config in `/etc/amdgpu-fan.yml`:
```
...
cards:
- card1
```
This will skip `card0`, and control fans on `card1` only

Akin2Silver, 2021-01-22:
Having the same issue on Manjaro. 
Is this project still maintained?

Akin2Silver, 2021-01-23:
I think i found the issue I have been trying to get it working and the issue is related to the write operation when writing to the pwm1 extension. 
I was reading this error https://gitlab.freedesktop.org/drm/amd/-/issues/1164
One person suggests that they are getting invalid argument when trying to write values below 3. quick test with my config file showed that to be the case. 

Edit the config file so the lowest value is 4 not 0 and my service is up and running again.

chestm007, 2025-02-28:
@LucasSymons  is this still a problem?

chestm007, 2025-03-02:
Fix merged in - closed. thanks!

closed 2025-03-02 by chestm007

LucasSymons, 2025-03-09:
> [@LucasSymons](https://github.com/LucasSymons) is this still a problem?

I am not sure sorry, I swapped away from this project.