chestm007/amdgpu-fan #19: OSError: [Errno 22] Invalid argument

codywohlers, 2020-11-05:
```
/usr/local/lib/python3.8/dist-packages/amdgpu_fan/controller.py:44: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  return yaml.load(f)
starting amdgpu-fan
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/amdgpu-fan", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/amdgpu_fan/controller.py", line 78, in main
    FanController(config).main()
  File "/usr/local/lib/python3.8/dist-packages/amdgpu_fan/controller.py", line 37, in main
    card.set_fan_speed(speed)
  File "/usr/local/lib/python3.8/dist-packages/amdgpu_fan/lib/amdgpu.py", line 77, in set_fan_speed
    return self.write_endpoint('pwm1', int(speed))
  File "/usr/local/lib/python3.8/dist-packages/amdgpu_fan/lib/amdgpu.py", line 42, in write_endpoint
    return e.write(str(data))
OSError: [Errno 22] Invalid argument
```

codywohlers, 2025-04-25:
I still have that system so I'll give it a try when I get a chance.

doncato, 2025-04-30:
> is this still a problem?

Hi, just chiming in to mention I have the same (or a very similar?) problem.

My Error output:
````yaml
starting amdgpu-fan
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/amdgpu-fan", line 8, in <module>
    sys.exit(main())
             ~~~~^^
  File "/usr/lib/python3.13/site-packages/amdgpu_fan/controller.py", line 95, in main
    FanController(config).main()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/lib/python3.13/site-packages/amdgpu_fan/controller.py", line 49, in main
    card.set_fan_speed(speed)
    ~~~~~~~~~~~~~~~~~~^^^^^^^
  File "/usr/lib/python3.13/site-packages/amdgpu_fan/lib/amdgpu.py", line 77, in set_fan_speed
    return self.write_endpoint('pwm1', int(speed))
           ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/amdgpu_fan/lib/amdgpu.py", line 41, in write_endpoint
    with open(self._endpoints[endpoint], 'w') as e:
         ~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument
````

My config:

````yaml
#Fan Control Matrix. [<Temp in C>,<Fanspeed in %>]
speed_matrix:
- [0, 0]
- [30, 33]
- [45, 50]
- [60, 66]
- [65, 69]
- [70, 75]
- [75, 89]
- [80, 100]

# optional
cards:  # can be any card returned from `ls /sys/class/drm | grep "^card[[:digit:]]$"`
- card1

# optional
# temp_drop: 5  # how much temperature should drop before fan speed is decreased
````

````sh
doncato ~ $ ls /sys/class/drm -1    
card0@
card0-DP-4@
card0-HDMI-A-2@
card0-Writeback-2@
card1@
card1-DP-1@
card1-DP-2@
card1-DP-3@
card1-HDMI-A-1@
card1-Writeback-1@
renderD128@
renderD129@
version
````

If wanted I am willing to provide further information