chestm007/amdgpu-fan #26: amdgpu-fan wont start

MatiasPujado, 2022-02-15:
### The Issue
I recently installed Debian Bullseye on my system, following this guide I found out about **amdgpu-fan**: `https://wiki.debian.org/AtiHowTo`. Once installed it began to run smoothly but today it has stopped and I dont know why. I'm not a Python savvy developer, Im still learning the language... so Im not sure how to fix this without making a mess. Hopefully someone will; here are the files I've created: 
**/etc/amdgpu-fan.yml**
```
speed_matrix: # -[temp(*C), speed(0-100%)]
- [0, 0]
- [30, 15]
- [50, 30]
- [70, 70]
- [90, 100]

# Optional
# cards: # Any card returned from: `ls /sys/class/drm | gep "^card[[:digit:]]$"`
# - card0
``` 
**/etc/systemd/system/amdgpu-fan.service**
```
[Unit]
Description=Service for controlling amdgpu fan.
After=syslog.service

[Service]
Type=simple
User=root
ExecStart=/usr/local/bin/amdgpu-fan
Restart=always

[Install]
WantedBy=multi-user.target
```
When I run either of these commands: `amdgpu-fan` / `sudo amdgpu-fan`  I get this output:
```
Traceback (most recent call last):
  File "/usr/local/bin/amdgpu-fan", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/amdgpu_fan/controller.py", line 67, in main
    config = load_config(location)
  File "/usr/local/lib/python3.9/dist-packages/amdgpu_fan/controller.py", line 44, in load_config
    return yaml.load(f)
TypeError: load() missing 1 required positional argument: 'Loader'
``` 
And this is the output of: `sudo systemctl status amdgpu-fan.service`
```
● amdgpu-fan.service - Service for controlling amdgpu fan.
     Loaded: loaded (/etc/systemd/system/amdgpu-fan.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2022-02-15 19:40:26 -03; 16min ago
    Process: 699182 ExecStart=/usr/local/bin/amdgpu-fan (code=exited, status=1/FAILURE)
   Main PID: 699182 (code=exited, status=1/FAILURE)
        CPU: 652ms
```
### What I've tried
Uninstalling **amdgpu-fan** and deleting the 2 files I've created.
Then I reinstalled **amdgpu-fan** and made again the 2 files mentioned above.

I'd appreciate if anyone could point me in the right direction. Thanks in advance.

ZyxerZyxani, 2022-03-01:
https://github.com/chestm007/amdgpu-fan/commit/b0cb2d0da866c58329ce2edf146be1c41d3f6d8d

You need to change that line in the controller.py file. I got exactly same error but changing that fixed it

ZyxerZyxani, 2022-03-01:
Temporary fix, Edit /usr/local/lib/python3.9/dist-packages/amdgpu_fan/controller.py
and change yaml.load(f) to yaml.safe_load(f)

MatiasPujado, 2022-03-01:
Thank you Zyxer! that did the trick.

closed 2022-03-01 by MatiasPujado