chestm007/amdgpu-fan #27: sudo pip install amdgpu-fan old code

ZyxerZyxani, 2022-03-01:
The pip package/library when installed uses old code that has not changed the yaml.load(f) part to yaml.safe_load(f), thus all installs using new libraries will get error when trying to run sudo amdgpu-fan after sudo pip3 install amdgpu-fan.

yaml.load(f) expects another argument in newer version, and thus throws error.
```
Traceback (most recent call last):
  File "/usr/bin/amdgpu-fan", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.10/site-packages/amdgpu_fan/controller.py", line 67, in main
    config = load_config(location)
  File "/usr/lib/python3.10/site-packages/amdgpu_fan/controller.py", line 44, in load_config
    return yaml.load(f)
TypeError: load() missing 1 required positional argument: 'Loader'
```

Changing that line to yaml.safe_load(f) made it work for me

akrenz, 2022-06-28:
Not just installing the package fails, but you also cannot run amdgpu-fan on systems with an updated package. Tested this on openSUSE Tumbleweed 20220623. changing `yaml.load(f)` to `yaml.safe_load(f)` also fixed it for me.

MekayelAnik, 2024-03-20:
After solving the issue by changing `yaml.load(f)` to `yaml.safe_load(f)`, the issue reappears.

MekayelAnik, 2025-02-28:
Yes this issue still exists.

chestm007, 2025-03-01:
this issue should now be fixed in version 0.0.23+, just pushed to pypi

closed 2025-03-01 by chestm007