chestm007/amdgpu-fan #28: Add openrc support

supasonix-zuv, 2022-04-10:


supasonix-zuv, 2022-04-10:
whoops, accidentally commented before I was done. 
Anyway, it'd be cool if other OpenRC users could run this without having to write their own init file. I wrote up one, but I've never done it before, so IDK if it's any good.
[amdgpu-fan.txt](https://github.com/chestm007/amdgpu-fan/files/8459126/amdgpu-fan.txt)
I also wrote the config file, cause I think openrc wants it even if it doesn't have any settings, but IDK.
[amdgpu-fanconf.txt](https://github.com/chestm007/amdgpu-fan/files/8459128/amdgpu-fanconf.txt)

chestm007, 2025-02-28:
I'm happy for someone to take the lead on this and submit a PR - im not an open RC user