chestm007/amdgpu-fan #30: missing 1 required positional argument: 'Loader'

daniel712, 2022-07-08:
unfortunately not running here an debian bullseye and AMD Navi (5700 XT):

Traceback (most recent call last):
  File "/usr/local/bin/amdgpu-fan", line 8, in <module>   sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/amdgpu_fan/controller.py", line 76, in main   config = ad_config(CONFIG_LOCATIONS[-1])
  File "/usr/local/lib/python3.9/dist-packages/amdgpu_fan/controller.py", line 44, in load_config   return yaml.load(f)
TypeError: load() missing 1 required positional argument: 'Loader'

19721dan, 2022-10-30:
same problem. always worked before. mxlinux 21

codesnippy, 2023-01-27:
Same issue here Also Debian Bullseye

jacness12334, 2023-03-06:
Same issue, debian bullseye.

sanwill, 2023-03-21:
The amdgpu-fan package at  pip is not updated with this fix.
The error was resolved after apply the fix, tested on Fedora and Ubuntu.

chestm007, 2025-02-28:
pip package needs to be updated

closed 2025-02-28 by chestm007