chestm007/amdgpu-fan #33: KeyError: 'pwm1_min'

YuruiHong, 2023-11-24:
Installed the lasted version from GitHub, but got this:

```python
starting amdgpu-fan
Traceback (most recent call last):
  File "/usr/local/bin/amdgpu-fan", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/amdgpu_fan/controller.py", line 78, in main
    FanController(config).main()
  File "/usr/local/lib/python3.11/dist-packages/amdgpu_fan/controller.py", line 35, in main
    logger.debug(f'{name}: Temp {temp}, Setting fan speed to: {speed}, fan speed{card.fan_speed}, min:{card.fan_min}, max:{card.fan_max}')
                                                                                                       ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/amdgpu_fan/lib/amdgpu.py", line 64, in fan_min
    return int(self.read_endpoint('pwm1_min'))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/amdgpu_fan/lib/amdgpu.py", line 36, in read_endpoint
    with open(self._endpoints[endpoint], 'r') as e:
              ~~~~~~~~~~~~~~~^^^^^^^^^^
KeyError: 'pwm1_min'
```

Also tried all the pull request versions and got the same error.🌚

visionscaper, 2024-03-22:
I also have the exact same error. @chestm007 Any idea what could be the problem here? Thanks.

RGD2, 2024-06-20:
I believe this is because the code assumes all cards support that item in the `/sys/class/drm/card0/device/hwmon/hwmon2/` directory. I get this crash because I've got built-in graphics showing up in `/sys/class/drm/card1/device/hwmon/hwmon3/`, and  there's no `pwm1_min` endpoint in that directory, hence the `KeyError`.

Code needs a change to fail gracefully in such cases - preferably to ignore cards which don't have the right endpoints.
Workaround might be to disable the on-CPU graphics so it doesn't show up to confuse amdgpu-fan.

RGD2, 2024-06-20:
Better workaround:
append this to your `/etc/amdgpu-fan.yml` file:
```
cards:
- card0
```

This is enough to make it ignore card1, the built-in graphics. Solved the issue for me immediately.

chestm007, 2025-03-02:
@YuruiHong @RGD2 @visionscaper @gokaybiz 
Could you guys please test the changes in the PR and let me know if it fixes the problem?

ahmaddie1378, 2025-04-04:
still same bug , i using yay version