chestm007/amdgpu-fan #34: Service Error - OSError: [Errno 95] Operation not supported

r00ms, 2024-01-23:
Hi, I am trying to get amdgpu-fan working due to high GPU temps, I followed the steps outlined on the [Debian Wiki](https://wiki.debian.org/AtiHowTo). 

When checking the status of the service after restarting the daemon and system entirely I find this:

```
× amdgpu-fan.service - Service for controling amdgpu fan.
     Loaded: loaded (/etc/systemd/system/amdgpu-fan.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2024-01-23 19:04:12 GMT; 2s ago
   Duration: 107ms
    Process: 6776 ExecStart=/usr/local/bin/amdgpu-fan (code=exited, status=1/FAILURE)
   Main PID: 6776 (code=exited, status=1/FAILURE)
        CPU: 811ms

Jan 23 19:04:12 Computer amdgpu-fan[6776]:                                                                             >
Jan 23 19:04:12 Computer amdgpu-fan[6776]:   File "/usr/local/lib/python3.11/dist-packages/amdgpu_fan/lib/amdgpu.py", l>
Jan 23 19:04:12 Computer amdgpu-fan[6776]:     return int(self.read_endpoint('fan1_input'))
Jan 23 19:04:12 Computer amdgpu-fan[6776]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jan 23 19:04:12 Computer amdgpu-fan[6776]:   File "/usr/local/lib/python3.11/dist-packages/amdgpu_fan/lib/amdgpu.py", l>
Jan 23 19:04:12 Computer amdgpu-fan[6776]:     return e.read()
Jan 23 19:04:12 Computer amdgpu-fan[6776]:            ^^^^^^^^
Jan 23 19:04:12 Computer amdgpu-fan[6776]: OSError: [Errno 95] Operation not supported
Jan 23 19:04:12 Computer systemd[1]: amdgpu-fan.service: Main process exited, code=exited, status=1/FAILURE
Jan 23 19:04:12 Computer systemd[1]: amdgpu-fan.service: Failed with result 'exit-code'.
```

Which is odd because I actually have managed to get the service to run, maybe this was with the config file not requiring modifications to the pwm1 file?

```
● amdgpu-fan.service - Service for controling amdgpu fan.
     Loaded: loaded (/etc/systemd/system/amdgpu-fan.service; enabled; preset: enabled)
     Active: active (running) since Tue 2024-01-23 18:45:51 GMT; 8ms ago
   Main PID: 4461 (amdgpu-fan)
      Tasks: 1 (limit: 18997)
     Memory: 2.5M
        CPU: 3ms
     CGroup: /system.slice/amdgpu-fan.service
             └─4461 /usr/bin/python3 /usr/local/bin/amdgpu-fan
```

I have inspected the /usr/local/lib/python3.11/dist-packages/amdgpu_fan/lib/amdgpu.py file myself but cannot seem to find any issues with it, other than the fact that these variables aren't pointing to the correct locations?

```
ROOT_DIR = '/sys/class/drm'
HWMON_DIR = 'device/hwmon'
```

My systemd service looks like this:
                                                               
```
[Unit]
Description=Service for controling amdgpu fan.
After=syslog.service

[Service]
Type=simple
User=root
ExecStart=/usr/local/bin/amdgpu-fan
Restart=always

[Install]
WantedBy=multi-user.target
```

As mentioned I have been able to get the service to start but it has not modified the fan speeds for whatever reason (possibly due to the config file and current temps) **but** I have been able to use a different tool to control the fan speed so I know that it definitely can be done, I would just rather rely on this tool rather than that one. 

**The GPU in question**
```
08:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Pitcairn XT [Radeon HD 7870 GHz Edition]
	Subsystem: Micro-Star International Co., Ltd. [MSI] Pitcairn XT [Radeon HD 7870 GHz Edition]
	Kernel driver in use: amdgpu
	Kernel modules: radeon, amdgpu

```

As far as I know, the card (although old) should be supported by both amdgpu and amdgpu-fan. 

Any help is much appreciated to save my GPU from frying.

Thanks in advance!

chestm007, 2025-02-28:
can you run amdgpu-fan from a terminal and paste the output in here please? :)
also, you mentioned ROOT_DIR and HWMON_DIR are wrong, what should they be on your system

rpenido, 2025-03-01:
I have the same issue.
This is the output for the amdgpu-fan command:

```
starting amdgpu-fan
Traceback (most recent call last):
  File "/bin/amdgpu-fan", line 8, in <module>
    sys.exit(main())
             ~~~~^^
  File "/usr/lib/python3.13/site-packages/amdgpu_fan/controller.py", line 95, in main
    FanController(config).main()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/lib/python3.13/site-packages/amdgpu_fan/controller.py", line 39, in main
    temp = card.gpu_temp
           ^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/amdgpu_fan/lib/amdgpu.py", line 56, in gpu_temp
    return float(self.read_endpoint('temp1_input')) / 1000
                 ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/amdgpu_fan/lib/amdgpu.py", line 37, in read_endpoint
    return e.read()
           ~~~~~~^^
PermissionError: [Errno 1] Operation not permitted
```

chestm007, 2025-03-02:
try running with sudo, sometimes the files it needs to read are root only.
Also, what distro are you on and can you paste in your config? 
Cheers!

rpenido, 2025-03-15:
Sorry for the delay! 😄
This log is from running with sudo.

This is my config.

```yml
#Fan Control Matrix. [<Temp in C>,<Fanspeed in %>]
speed_matrix:
- [0, 0]
- [30, 33]
- [45, 50]
- [60, 66]
- [65, 69]
- [70, 75]
- [75, 89]
- [80, 100]

# optional
# cards:  # can be any card returned from `ls /sys/class/drm | grep "^card[[:digit:]]$"`
# - card1

# optional
# temp_drop: 5  # how much temperature should drop before fan speed is decreased
```

I'm using arch (`6.13.6-arch1-1`), and I have an eGPU.
It's not working right now, but I think it is intermittent.

Let me know if I can help with something!

chestm007, 2025-03-19:
hmmmmm.

try setting the following in your config.
```
 cards:
 - card1
```

If that doesnt work, please let me know the output of
`ls /sys/class/drm`

Cheers!

rpenido, 2025-03-19:
Thank you for the support @chestm007!

> try setting the following in your config.

This results in `no compatible cards found, exiting`. When I change it to "card2", I get the previous error.

> If that doesnt work, please let me know the output of
```
card1       card1-DP-2  card1-DP-4  card1-eDP-1     card2        card2-DP-11  card2-HDMI-A-3     renderD128  version
card1-DP-1  card1-DP-3  card1-DP-5  card1-HDMI-A-1  card2-DP-10  card2-DP-9   card2-Writeback-2  renderD130
```

As I said, the issue is intermittent (on restarts, I think). When it works again, I will try to gather more information.

chestm007, 2025-06-11:
> Thank you for the support @chestm007!
>
>
> This results in `no compatible cards found, exiting`. When I change it to "card2", I get the previous error.
>
> ```
> card1       card1-DP-2  card1-DP-4 [...]

Can you list the files under

/sys/class/drm/card1/hwmon/hwmon*/
/sys/class/drm/card2/hwmon/hwmon*/

You may need to change the * to numbers appliccable in the parent directory.